xinkaiwang/rpio-pwm #1: Working with two different PWM frequency ?

Serveurperso, 2018-09-26:
Hello,

I need to drive Servo - 20ms standard repeat time - and H-Bridge with faster repeat time (3000 uS)

I tested with two (unused) different DMA channel but the slower PWM (servo) go wrong.

I also tested to use the hardware PWM library "raspi-pwm" but the DMA PWM "rpio-pwm" stop working after writing to HPWM I/O.

Any idea ?

xinkaiwang, 2020-03-13:
Due to those 2 application require 2 different cycle time, I believe you will just need to use 2 different DMA channels for those 2 applications. I use channel 14 in my demo, you may need to do some homework to find out another un-used channel for you to use.

closed 2020-03-13 by xinkaiwang

xinkaiwang, 2020-03-23:
@Serveurperso , you may want to try my newest update 1.0.x which added support for the mode you need. And you may also want to check my other project perf-rpio, where you can find sample code on how to setup 2 channel to do those things at the same time.