xinkaiwang/rpio-pwm #2: Can't npm install on PI ZERO : asyncWrite is not a function

Serveurperso, 2018-09-26:
I use the same minimalist image on all my RPI 2/3B+
All work but not the PI Zero (same image but with the good architecture).

I tested some node version (n) stable & lts :

    node/10.8.0
    node/10.10.0
  ο node/8.12.0

I always get :

```
(root|/var/www) sudo npm cache clean -f
npm WARN using --force I sure hope you know what you are doing.
(root|/var/www) npm install rpio-pwm
npm ERR! asyncWrite is not a function
npm ERR! pna.nextTick is not a function

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2018-09-26T18_29_00_032Z-debug.log
(root|/var/www) cat /root/.npm/_logs/2018-09-26T18_29_00_032Z-debug.log
0 info it worked if it ends with ok
1 verbose cli [ '/usr/local/bin/node',
1 verbose cli   '/usr/local/bin/npm',
1 verbose cli   'install',
1 verbose cli   'rpio-pwm' ]
2 info using npm@6.2.0
3 info using node@v10.8.0
4 verbose npm-session bb9531f6a7f2a6f0
5 silly install loadCurrentTree
6 silly install readLocalPackageData
7 verbose stack TypeError: asyncWrite is not a function
7 verbose stack     at onwrite (/usr/local/lib/node_modules/npm/node_modules/readable-stream/lib/_stream_writable.js:480:7)
7 verbose stack     at WritableState.onwrite (/usr/local/lib/node_modules/npm/node_modules/readable-stream/lib/_stream_writable.js:180:5)
7 verbose stack     at WriteStream.to [as _worker] (/usr/local/lib/node_modules/npm/node_modules/pacote/node_modules/make-fetch-happen/cache.js:154:13)
7 verbose stack     at WriteStream._write (/usr/local/lib/node_modules/npm/node_modules/mississippi/node_modules/flush-write-stream/index.js:35:13)
7 verbose stack     at doWrite (/usr/local/lib/node_modules/npm/node_modules/readable-stream/lib/_stream_writable.js:428:64)
7 verbose stack     at writeOrBuffer (/usr/local/lib/node_modules/npm/node_modules/readable-stream/lib/_stream_writable.js:417:5)
7 verbose stack     at WriteStream.Writable.write (/usr/local/lib/node_modules/npm/node_modules/readable-stream/lib/_stream_writable.js:334:11)
7 verbose stack     at WriteStream.to [as _worker] (/usr/local/lib/node_modules/npm/node_modules/pacote/node_modules/make-fetch-happen/cache.js:171:25)
7 verbose stack     at WriteStream._write (/usr/local/lib/node_modules/npm/node_modules/mississippi/node_modules/flush-write-stream/index.js:35:13)
7 verbose stack     at doWrite (/usr/local/lib/node_modules/npm/node_modules/readable-stream/lib/_stream_writable.js:428:64)
7 verbose stack     at writeOrBuffer (/usr/local/lib/node_modules/npm/node_modules/readable-stream/lib/_stream_writable.js:417:5)
7 verbose stack     at WriteStream.Writable.write (/usr/local/lib/node_modules/npm/node_modules/readable-stream/lib/_stream_writable.js:334:11)
7 verbose stack     at WriteStream.to [as _worker] (/usr/local/lib/node_modules/npm/node_modules/pacote/node_modules/make-fetch-happen/cache.js:182:19)
7 verbose stack     at WriteStream._write (/usr/local/lib/node_modules/npm/node_modules/mississippi/node_modules/flush-write-stream/index.js:35:13)
7 verbose stack     at doWrite (/usr/local/lib/node_modules/npm/node_modules/readable-stream/lib/_stream_writable.js:428:64)
7 verbose stack     at writeOrBuffer (/usr/local/lib/node_modules/npm/node_modules/readable-stream/lib/_stream_writable.js:417:5)
8 verbose cwd /var/www
9 verbose Linux 4.14.70+
10 verbose argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "rpio-pwm"
11 verbose node v10.8.0
12 verbose npm  v6.2.0
13 error asyncWrite is not a function
14 verbose exit [ 1, true ]

```

Serveurperso, 2018-09-26:
I found a workaround :
I must rewind to the node version 8.9.4 and install work !

xinkaiwang, 2020-03-10:
I'm glad you found a solution. Seems the recent version of node.js do have more and more breaking changes. So my-self is still using v6.xx for my projects.

closed 2020-03-10 by xinkaiwang