xinkaiwang/rpio-pwm #5: control SK6812 (LED strip)

UrielCh, 2020-06-14:
Hi,

I would like to control some LED strip in nodeJS.

like in https://github.com/jgarff/rpi_ws281x

I opened a pull request with a forked version than contains typing and works on all current nodejs version.

Can you give me some tips about PWM on raspberry Pi ?

Should your API give write access to DMA memory block to fill a buffer containing all the data I want to send to my PWM pin at a given frequency ?

Without an oscilloscope I'm blind.

xinkaiwang, 2020-08-16:
`Without an oscilloscope I'm blind` I can feel your pain. That's exactly why I'm building this lib here. Ideally, all those low-level dirty details (such access to DMA memory block, defensive coding for bulletproof  safe) should be handled by a well encapsulated as a black-box, and no one else need to worry about it ever after.  
My current rpio-pwm supports creating up to 32 pins per channel, all those pins to drive your LED. I'm not clear which function is still missing from your point of view.

xinkaiwang, 2020-08-16:
I see what you want is using nodejs to control ws281x. In that case probably you can just implement another nodejs wrapper for rpi_ws281x. The core part of rpi_ws281x is implemented in c. you can use my project as an example on how to wrap c code into nodejs module.

UrielCh, 2020-08-17:
Hi, I last I made it works using thius code:
https://github.com/UrielCh/rpi-ws281x

Making it work's on a Pi Zero was the worse part. this board is soooooooo slooow ...

UrielCh, 2020-08-17:
The code I PR to you should contains good stuff, even if I did not used it.

closed 2021-09-12 by xinkaiwang